dparrini/python-comtrade #2: Issues with type conversation

mathiasrabe, 2019-04-18:
I have some issues with the following lines:

                    skew = float(skew)
                    cmin = int(cmin)
                    cmax = int(cmax)

Python detects skew as string, so the type conversation fails. It seems that skew is empty in my measurement file.

cmin and cmax are floating point variables. Type conversation fails as well on these two lines.

I changed it to the following lines but I think there should be a better solution:

                    try:
                       skew = float(skew)
                    except:
                       pass
                    cmin = float(cmin)
                    cmax = float(cmax)

dparrini, 2019-04-18:
Hi Mathias, thank you for your comment and feedback. The IEEE COMTRADE 2013 standard specifies the `skew` field as a critical field, i.e., that must have non-null value. Since there are lots of comtrade files provided by different vendors that doesn't follow 100% any standard, it's hard to guess which fields may be left in blank or may have different than expected values.

Even  though I try to make the script follow the standard, it eventually will fail for "real world" files, and this is one case. I think for the `skew` case it's ok to set it's value to 0 when it's null. The old 1991 standard even suggest 0 as a default value in one of it's examples. Your solution is good, but instead I used an if statement and set it to zero when it's an empty string. Also, I did that to the `b` value, since it's frequently zero. 

I think a better solution is a fail-safe reader, one that is loose about the standard and assumes that the file may not have all the values set. I'll think about that. Out of curiosity, what's the year specified in your .cfg file?

About cmin and cmax, they refer to the `min` and `max` fields, and you're right! By the standard (2013) they can be `int` or `float` values, so your solution is a good one. By mistake I considered the 1991 standard, which defines then as integers.

Thank you for your contributions!

mathiasrabe, 2019-04-18:
Thanks for you fast answer and fixes, David. Due to holidays I can't answer your questions at the moment because the files are saved on my companies computer. I will write you as soon as possible.

I'm not very familiar with the COMTRADE standard. I just searched for a library that can read some files a colleague of mine gave me and I found your git repo. Thank you for that piece of code! Very handy.

mathiasrabe, 2019-04-23:
Here is my .cfg file. I changed the `<filename>` for anonymity.

```
<filename>,,1999
1,1A,0D
1,Spg,,,V,0.787112295761296,-46.6318359375,,-25837.546875,25744.283203125,1,1,p
50
1
200000,43914000
14/06/2018,09:31:49.000694
14/06/2018,09:31:49.000694
Binary
1
```
I also have a .inf file. There you can see that the source of this file is DEWESoft 7.1.2

```
[Public Record_Information]
[Public Event_Information_#1]
Source=DEWESoft 7.1.2 Datei
[Public Event_Information_#1]
Channel_number=1
max_value=25744.283203125
min_value=-25837.546875
[Public File_Description]
Total_Channel_Count=1
Analog_Channel_Count=1
Sample_Rate=200000
File_Start_Time=14/06/2018,09:31:49.000694
Trigger_Time=14/06/2018,09:31:49.000694
File_Type=binary
[Public Analog_Channel_#1]
Channel_ID=Spg
Channel_Units=V
Channel_Multiplier=0.787112295761296
Channel_Offset=-46.6318359375
[Company1 event_rec]
[Company1 analog_rec_1]
```

dparrini, 2019-04-29:
Hi Mathias. I'm glad the script helped you, I'm also happy that it worked well with such large file. I don't have test files originated from data loggers, only protective relays and fault recorders, so thank you for testing the script.

The current version don't support .inf and .hdr file, sadly, but for the former I think you can use a .INF file parser, such as the recommended in the link below. The COMTRADE INF format is not different from a Windows configuration file, so they might be useful.

https://stackoverflow.com/questions/927279/how-to-parse-windows-inf-files-for-python

I'm closing the issue.

closed 2019-04-29 by dparrini

mathiasrabe, 2019-05-01:
> I'm also happy that it worked well with such large file.

It is working but it is not very fast. It took about 170 seconds for my file on an i7 9700(?). The problem is this while-loop I think: https://github.com/dparrini/python-comtrade/blob/504466ba4c65cf60f1b07cf961b639f9970e3964/comtrade.py#L703 

> The current version don't support .inf and .hdr file

Just because I'm curious: Is it planed to support these files in a future release?

>  but for the former I think you can use a .INF file parser, such as the recommended in the link below.

Thanks, I will try it.

But nevertheless it's a very handy piece of code. Thank you for that!

dparrini, 2019-05-01:
Hi Mathias,

I pushed a new commit that improves binary data reading. In a test I did, it read a file 3 times faster.

You're right about where the code was slow. The previous version used multiple calls to `struct.unpack`, inside the while loop, which was inefficient. The solution was to instantiate a `struct.Struct` object and to call it's `unpack` method once for each row of data. For files with multiple channels, the improvement is even more apparent.

> Just because I'm curious: Is it planed to support these files in a future release?

In the last commit I added support for binary32 and float32 files, which I was working on. In parallel I'm working to add methods to better deal with time information and to write Comtrade files based on objects created by the user. Honestly, I didn't plan for supporting them soon because they are files with common formats that other tools may help the users out. But since they are in the standard, I should start working on them too.

mathiasrabe, 2019-05-10:
> I pushed a new commit that improves binary data reading. In a test I did, it read a file 3 times faster.

Indeed. It reduced the execution time to 62 seconds in my case. Excellent work!