dparrini/python-comtrade #3: Wrong time reference in BinaryDatReader

mathiasrabe, 2019-07-15:
Each timestep is calculated by the grid frequency and not by the sample frequency:

https://github.com/dparrini/python-comtrade/blob/30dba39299d48ac8a602967d848c426d3deaaff5/comtrade.py#L953

In my example file the frequency variable will be 50, like 50 Hz, but the sample frequency is 200 kHz. So the resulting time delta in each column will be 0.02 sec which is not correct.

dparrini, 2019-07-15:
Hi @mathiasrabe , thank you for your feedback. I fixed this in `SamplingRates` branch, where I add support for multiple sampling rates. I'll merge this branch into master to fix this issue.

In this branch I add a `DatReader._get_time` method where it choose to use the dat file timestamp or one calculated using the sample frequency. It will only use the sampling frequency if the timestamp is missing.

dparrini, 2019-07-15:
I just saw your second edit, I will check this out.

> EDIT: I also checked the variable self._cfg.sample_rates and I got these values:
> [[200000.0, 43914000]]
> Don't know where the 43914000 comes from. The cfg file looks like this:
`Cfg.sample_rates` stores a list of a pair of values: the sample rate (`200000.0`) and the end sample (`43914000`) where the sample rate is applied. For a file with only one sample rate, the end sample will be the total samples of the file.

closed 2023-09-10 by dparrini