dparrini/python-comtrade #8: error: unpack requires a buffer of 100 bytes

maxwickPPL, 2020-08-05:
can't read my .cfg and .dat files,  I'm using the code rather than install

dparrini, 2020-08-05:
Hi @maxwickPPL , could you provide a code snippted with how you are using the code? Thanks

maxwickPPL, 2020-08-05:
import matplotlib.pyplot as plt
I use the whole comtrade.py file in the cell above, the I run below calling the class instead of a module.  

rec = Comtrade()
rec.load("HR_11435.cfg", "HR_11435.dat")
print("Trigger time = {}s".format(rec.trigger_time))

plt.figure()
plt.plot(rec.time, rec.analog[0])
plt.plot(rec.time, rec.analog[1])
plt.legend([rec.analog_channel_ids[0], rec.analog_channel_ids[1]])
plt.show()

---------------------------------------------------------------------------
error                                     Traceback (most recent call last)
<ipython-input-7-f7e9cc3c3ab9> in <module>
      3 
      4 rec = Comtrade()
----> 5 rec.load("HR_11435.cfg", "HR_11435.dat")
      6 print("Trigger time = {}s".format(rec.trigger_time))
      7 

<ipython-input-1-c4ee9e70e94e> in load(self, cfg_file, dat_file, **kwargs)
    691 
    692             # load both cfg and dat
--> 693             self._load_cfg_dat(cfg_file, dat_file)
    694 
    695             # Load additional inf and hdr files, if they exist.

<ipython-input-1-c4ee9e70e94e> in _load_cfg_dat(self, cfg_filepath, dat_filepath)
    713 
    714         dat = self._get_dat_reader()
--> 715         dat.load(dat_filepath, self._cfg)
    716 
    717         # copy dat object information

<ipython-input-1-c4ee9e70e94e> in load(self, dat_filepath, cfg)
    888             self._preallocate()
    889             with open(self.file_path, self.read_mode) as contents:
--> 890                 self.parse(contents)
    891         else:
    892             raise FileNotFoundError(errno.ENOENT, os.strerror(errno.ENOENT),

<ipython-input-1-c4ee9e70e94e> in parse(self, contents)
   1075         irow = 0
   1076         while row != b'':
-> 1077             values = row_reader.unpack(row)
   1078             # Sample number
   1079             n = values[0]

dparrini, 2020-08-05:
Ok, the script looks good. Do you have any details about the comtrade file you are trying to read? if you could not share the cfg file, please share the first 2 lines and the last 7 lines of the file. Example:

```station,equipment,1999
20,4A,16D
...
60.000000000
1
15360.000000000,5
07/01/2017,15:35:41.958268
07/01/2017,15:35:41.958333
BINARY
1
```

maxwickPPL, 2020-09-02:
STATIONLAB,FID=SEL-487E-X-XXXX-XX-XXXXXXX-XXXXXXXXX,1999
361,25A,336D
60
1
2000,1000
31/08/2020,21:59:59.903096
31/08/2020,22:00:00.004429
BINARY
1

maxwickPPL, 2020-09-02:
these are test relays in our lab,  so config file may be weird, I opened the cfg with visual studio

maxwickPPL, 2020-09-09:
Is my cfg wrong?

dparrini, 2020-10-27:
Hi @maxwickPPL the provided CFG doesn't contains channel information. I created fake channels and data to test the part you provided and couldn't reproduce the problem. Also I don't see a problem copying and using the module contents, it should work as well.

Maybe information about you environment can help. Which Python version are you using? Do you have a pair of cfg/dat files that have this problem that you could share?

closed 2023-09-10 by dparrini