dparrini/python-comtrade #9: 'NoneType' object has no attribute 'group' while opening an SEL 735 PQ meter file

spark706, 2020-11-11:
I am writing a conversion tool to open 1999 Binary and 2013 Binary32 versions of high resolution COMTRADE data.  I have no issues with the 2013 version but I am having one with the 1999.  In looking at the error (image) I get reading the 1999 version regarding timestamp errors and seeing the supported capabilities in the Readme file, it appears like this format is accepted.

[image]

Is there any known workaround within the COMTRADE reader for such an error seen? Unfortunately I cannot provide a copy of the file or even a screen shot of the file data as it comes from a utility.  I can say that there format has not been modified from standard as I am capable of opening the file in SEL SynchroWAVE and PQDiff with no issues.

I have been staring at the COMTRADE reader to determine is I could preprocess the .cfg file and wither change or remove the time stamp altogether as I don't need them for the particular project I and converting them for.  I simply need a list of voltage and currents, and possibly harmonics, in the time stamp order to import to Simulink.

dparrini, 2020-11-12:
Hi @spark706! In cases like this I ask for annonymized data such as the .cfg file with anything in its first line (which can describe the equipment and the station) and empty strings in place of channel names.

So, in this case I think the problem is the timestamp parsing. Could be an unsupported format or the module getting lost and trying to parse a string that is not a timestamp. Could you provide the annonymized cfg or the start and trigger timestamps? The last two should be enough if the problem is the timestamp format.

I noticed that the module doesn't handle well optional date and/or time in these timestamp fields, so I pushed an commit fixing that. The standard says these can be optional. Tell me if it works.

Thanks.

spark706, 2020-11-13:
I talked with an associate who solved the issue.  On line 56-58 there is a regex for timestamp and the 735 seems to be using a 3 decimal place as opposed to 5.  The update to the code which is now working is 

```
# timestamp regular expression
re_date = re.compile("([0-9]{1,2})/([0-9]{1,2})/([0-9]{2,4})")
re_time = re.compile("([0-9]{2}):([0-9]{2}):([0-9]{2})\\.([0-9]{3,12})")
```

The last bit was `{5,12})"` of re_time but by changing to `{3,12})"` as seen above also it is now working and still handles my 2013 version also.

the timestamp in the 1999 format is 
01/01/2020,06:22:35.108
01/01/2020,06:22:35.307

while the 2013 version is
01/01/2020,06:22:35.675637
01/01/2020,06:22:35.725636

dparrini, 2020-11-13:
Thank you @spark706 , this is the fraction of second timestamp part which the standard required the minimum of 6 characters length. I'll update to use 3 as your solution, but I think I'll make this fractional part optional in a future release. There's no benefit for the module being inflexible about this.

dparrini, 2020-11-13:
@spark706 I updated the module and also the public repository in PyPI (version 0.0.3) with an improved solution. 

In my solution, based on yours, the fraction of seconds part is optional, and timestamps like these are acceptable:

00:00:00
00:00:00.0
06:22:35.108
06:22:35.675637

one thing to notice is that the value after "." is used by the module as a number of microseconds to create a Python's `datetime.datetime` instance. So, for the `06:22:35.108` timestamp the correct handling should be to pad `108` with `0`s to the right: `108000`. The version I made available does that.

I'll close the issue if you have no remarks about it. Thank you for your input.

spark706, 2020-11-15:
Works for me.  Thanks for the help.

closed 2020-11-15 by spark706