dparrini/python-comtrade #10: Issues with Status (digital) channel information

mathiasrabe, 2020-11-24:
I have a several new contrade files. One is very interesting. Here are some lines of the cfg file:

```
x,,2001
146,2A,144D
1,<analog stuff not important here>
2,<analog stuff not important here>
1,Some Status Channel,,,
```

The last line is the issue. The following line in the code is not fault tolerant:
https://github.com/dparrini/python-comtrade/blob/78a4903fd6d1f3253097f18c8ff5b04be3c5ec62/comtrade.py#L399

You should add something like
```
            if not packed:
                # maybe throw some warning here
                packed = 0
```

dparrini, 2020-11-24:
Hi @mathiasrabe the `_read_sep_values` makes sure that the returned list contains at least `5` elements, filling the missing ones with `'0'`. It does nothing if the column values are empty. What I think is happening here is that the last column of the digital channel is critical and the code assumed so:

https://github.com/dparrini/python-comtrade/blob/78a4903fd6d1f3253097f18c8ff5b04be3c5ec62/comtrade.py#L403

I updated it to deal with an empty value without breaking. Please tell me if it's working. Thanks.

265273d2dbb650c6dde11193d3f868edfa1d42fe

mathiasrabe, 2020-11-24:
I've tested your commit and it's working. Thanks!

closed 2020-11-24 by mathiasrabe