dparrini/python-comtrade #11: Issues with type conversation (again)

mathiasrabe, 2020-11-24:
I have an issue with type conversation (again). I allready commented it in https://github.com/dparrini/python-comtrade/issues/2#issue-434918587

It seems that a new commit changed the type conversation back to this:

https://github.com/dparrini/python-comtrade/blob/78a4903fd6d1f3253097f18c8ff5b04be3c5ec62/comtrade.py#L387-L388

As you allready mentioned in https://github.com/dparrini/python-comtrade/issues/2#issue-434918587 the data type can be float too.

dparrini, 2020-11-24:
HI @mathiasrabe you're correct, I mentioned this, but I think in the previous issue I only fixed the `skew` problem (in that case `skew` could also be empty). I'm fixing this now.

dparrini, 2020-11-24:
Fixed with the last commit: 1778e90c845cdc707adddf2552ecc42930677bfb
Please tell me if its working for you.

mathiasrabe, 2020-11-24:
It's working. Thanks!

closed 2020-11-24 by mathiasrabe