dparrini/python-comtrade #12: UnicodeDecodeError: 'ascii' codec can't decode byte 0xd6 in position 0: ordinal not in range(128)

sjtuztq, 2020-11-29:
rec.load(fileCFG,fileDAT)

UnicodeDecodeError: 'ascii' codec can't decode byte 0xd6 in position 0: ordinal not in range(128)

dparrini, 2020-12-13:
Hi @sjtuztq , I pushed a commit addressing this issue.

a1026ad06e96eae0eb84814128fe03f5c5e93c75

Let me know if it worked for you. In case it don't fix the issue, please provide an anonymized version of the `fileCFG` file, keeping any special character there. Thanks

LFBJC, 2021-08-17:
@dparrini I have a similar issue, when using the version 0.0.4 it raises a "**UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe7 in position 7: invalid continuation byte**", so I had to downgrade to version 0.0.3 in order to read some of the oscillographies, but the version 0.0.3 doesn't read all oscillographies because of some issues with this version

dparrini, 2021-08-17:
Hi @LFBJC I'll take a look into this, thanks for bringing this up. I'm updating the code to handle issue #18  and hope I can fix this one as well.

dparrini, 2021-08-17:
@LFBJC I disabled utf-8 decoding in the most recent commit to fix #18 issue. Let me know if this works for you.

LFBJC, 2021-08-17:
I think something that may help is that in my case I could read these oscillographies on version 0.0.3, I think it's because of [this commit](https://github.com/dparrini/python-comtrade/commit/a1026ad06e96eae0eb84814128fe03f5c5e93c75) what issue where you trying to address when you did this?

LFBJC, 2021-08-18:
@dparrini still the same problem
[image]

LFBJC, 2021-08-18:
@dparrini I just noted that the version I installed from pip (0.0.4) doesn't have the most recent commits, but that's the last version on pip, how should I install it?

dparrini, 2021-08-18:
Hi @LFBJC , I pushed a new commit fb09c66012fe36c03c88cc2e7fbe797e48ad8a0a with what I think is a fix to the problem you're having. I didn't release a version in pip with this new commit yet but will if it solves your issue.
You can repeat the test you did by downloading the most recent comtrade.py file.

Let me know if it works. Thanks for your patience.
/David

LFBJC, 2021-08-19:
Hi @dparrini I tried again, it raised another exception, but I think you're getting there.
[image]

LFBJC, 2021-08-19:
I did a pull request

dparrini, 2021-08-20:
Great @LFBJC , thank you for your support on this!

LFBJC, 2021-08-20:
@dparrini no worries! tell me when you release a version in pip

dparrini, 2021-08-23:
@LFBJC just published it.

closed 2021-08-23 by dparrini