dparrini/python-comtrade #13: Mind if I work on contributing Sphinx Documentation in GitHub pages?

engineerjoe440, 2020-12-02:
Hi there @dparrini !!!

I'm about to embark on a project of my own to make a Python package to read and interpret SEL (Schweitzer Engineering Laboratories) Compressed Event records (CEV files). I want to try to unify the package's interface as much as possible with yours as it seems that you've pretty well established the best ins-and-outs for this sort of work. But along those lines, I want to identify all of the properties, methods, and more that you've created. I thought it might be helpful if there was a Sphinx-based GitHub-pages-hosted documentation site that shows all of the available properties and methods for each of your module's classes.

What do you think? Would you mind if I fork your project and work on getting the documentation working in GitHub pages before opening a pull request to get it back in with your work? Ultimately, I'd like to get the docs working, and then show you what it looks like to get your feedback and thoughts before we try merging the PR. What do you think?

P.S. Here's a shameless plug for the project that I'm getting started on: [`pycev`](https://github.com/engineerjoe440/pycev)

dparrini, 2020-12-13:
Hi @engineerjoe440 , sorry for the late answer. Feel free to do that. Since the project is in an early version (`0.0.x` on `pip`), I might change properties/classes names in newer versions, although not soon. So any input about getting things clear would be nice. Good work on your project.

dparrini, 2020-12-13:
I'll let this issue open as a way to answer any question that you have during your documentation work. There's some things I want to point out before you start:

1. The `comtrade.py` script is self contained, the majority of the users like to copy it to their projects and use it instead of downloading via `pip`. There are some classes and methods there not intended to be used outside the script (ie., private). So, I think some of them don't need to be thoroughly documented to help users, other than to help the few module maintainers.
2. I'm not sure if we can document all aspects of the COMTRADE files. Even though they are standardized, I don't know if it's a public standard (I think not). So, until we are sure, this module documentation cannot be a COMTRADE file documentation.

engineerjoe440, 2020-12-13:
Thanks for your willingness to accept contributions! 

I've forked, and thrown together a quick documentation site here: https://engineerjoe440.github.io/python-comtrade/

It's a very rough first pass, I haven't done anything to clean much up, but thought I'd share what it's looking like, so you could see, and make comments about changes. The colors, and page style are the defaults provided, so those can easily be mixed up if you'd like. and pretty much all of the "documentation" is pulled directly from the markdown README file, or the module itself. I've "added" very little. Let me know what you think, and if I've documented too much or too little, let me know!

closed 2026-10-21 by dparrini