dparrini/python-comtrade #16: struct.error: unpack requires a buffer of 192 bytes

nolanbell, 2021-07-14:
[Archive.zip](https://github.com/dparrini/python-comtrade/files/6813151/Archive.zip)

Please help

engineerjoe440, 2021-07-19:
Looks like I can reproduce the issue from #15 with the attached CFG/DAT files:

```
python3 test.py 
Traceback (most recent call last):
  File "test.py", line 5, in <module>
    rec.load("./comtrade-2013-binary32.CFG", "./comtrade-2013-binary32.DAT")
  File "/home/joestan/Downloads/comtrade.py", line 756, in load
    self._load_cfg_dat(cfg_file, dat_file)
  File "/home/joestan/Downloads/comtrade.py", line 769, in _load_cfg_dat
    self._cfg.load(cfg_filepath)
  File "/home/joestan/Downloads/comtrade.py", line 326, in load
    self._read_io(cfg)
  File "/home/joestan/Downloads/comtrade.py", line 436, in _read_io
    self._start_timestamp, nanosec = _read_timestamp(ts_str, self.ignore_warnings)
  File "/home/joestan/Downloads/comtrade.py", line 144, in _read_timestamp
    hour, minute, second, microsecond, \
TypeError: cannot unpack non-iterable NoneType object
```

That's using the standard example from [the python-comtrade README](https://github.com/dparrini/python-comtrade#cfg-and-dat-files-all-revisions) simply configured for the files as:

```python
import matplotlib.pyplot as plt
from comtrade import Comtrade

rec = Comtrade()
rec.load("./comtrade-2013-binary32.CFG", "./comtrade-2013-binary32.DAT")
print("Trigger time = {}s".format(rec.trigger_time))

plt.figure()
plt.plot(rec.time, rec.analog[0])
plt.plot(rec.time, rec.analog[1])
plt.legend([rec.analog_channel_ids[0], rec.analog_channel_ids[1]])
plt.show()
```

### Findings:
It appears that the COMTRADE file has some incorrect formatting in the timestamp, as it reports single digits for the hour:
[image]
According to the 2013 COMTRADE standard, the hour must be 2 digits long, thus, a 1-digit hour is an illegal value.

### Suggestions:
Since this record appears to be generated by an SEL-751A, I'd suggest reaching out to SEL's technical support to further investigate why the hours are being incorrectly represented. You can find your local support representative by visiting https://selinc.com/support

In the interim, you can get past *this* error by substituting a 1 for the 2 located here in `comtrade.py`:
[image]
However, that leads to the problem you've reported here! I'll do a little more digging to see if I can uncover what that's all about!

nolanbell, 2021-07-20:
Exactly! I found this issue here by doing exactly that for issue #15. Thank you for the quick responses.

engineerjoe440, 2021-07-20:
@nolanbell, of course! I'll hopefully have another chance tomorrow to look into that structure issue. It seems to be something to do with failing structure usage. Will have to look closer. If nothing else, hopefully I can help track down the error of the failure and @dparrini can help elaborate on what's actually happening. :)

dparrini, 2021-07-20:
Hi @engineerjoe440, great contribution! Thanks for your help.

Even though this Python module is based on the standards, not all equipment follows it strictly. Your solution will work for the case @nolanbell showed but not every other.

An improvement of the fix you propose is replacing the 58 line:
```
re_time = re.compile("([0-9]{2}):([0-9]{2}):([0-9]{2})(\\.([0-9]{1,12}))?")
```
with:
```
re_time = re.compile("([0-9]{1,2}):([0-9]{2}):([0-9]{2})(\\.([0-9]{1,12}))?")
```

I'm not sure how familiar you are with regular expressions, but this `A{x,y}` means "Match with A repeating x to y times". So, with this form, the expression will match with `1` or `10`, `11`, `12`...

(If you want to, you can send a pull request with this solution :) ).

engineerjoe440, 2021-07-20:
Hi @dparrini ! That's a very good point you raise!

I'm in something of a quarrel with myself on this topic. It seems reasonable, to me, to expect that records that are being parsed are generated according to the standards, so we could expect the exact form of timestamp that you've specified in the released package. Yet, your point is valid, it seems that not all equipment may perfectly adhere to the standards. Silly of me to get so caught up in this, isn't it?

I will open a PR to supply the suggested fix, as you propose. Hopefully then, I can further investigate the particulars of the structure error.

Thank you again, David!

engineerjoe440, 2021-07-22:
Okay, so I think I might be getting closer to the root of this issue. Though, that's doing a fair bit of reverse-engineering, so feel free to redirect me if I'm off-base!

This record contains 11 analog channels and 696 status channels as per the .CFG file:
[image]

Now, given that the COMTRADE standard defines the length formula to be:
![made using quicklatex: https://quicklatex.com/](https://quicklatex.com/cache3/a8/ql_65e2b1f5d7e10e2e760045fccff82da8_l3.png)

Taking that into context, and reviewing the `struct.Struct` formatting-string that's being created as: `LL 11l 44H` and examining the length of each row to be 140 bytes, it seems that both the DAT file and the structure of the formatting string are in alignment according to the number of bytes, but examining what the struct module provides for the required length, we do see 192 bytes (clearly not the 140 which we're expecting).
[image]

Now, upon further investigation with the struct module, I found this thread on Stack Overflow: https://stackoverflow.com/questions/42415983/struct-length-of-long-depending-on-byte-order/51453307

Specifically, it calls out the use of the `!` character leading the formatting string to use the standard struct byte-sizing instead of the machine's native size for long-types. Sure enough, that checks out!
[image]

All this to say... I think that I've found the root of *this* issue, and I'll open a PR to address it very soon!

engineerjoe440, 2021-07-22:
A little more investigation, and I realized that there is some logic to attempt negotiating exactly this challenge! 
[image]

However, note that I inserted a test print statement, and when running my little test script, I do not see that the particular message is printed. So I think that will need to be addressed.

engineerjoe440, 2021-08-25:
@dparrini: this issue can be closed.