dparrini/python-comtrade #18: binary32 cff-files

chrisdinho, 2021-08-03:
Hi,

I'm having a hard time using python-comtrade with binary32 cff-files.
One problem might be, that the binary data can't be decoded in utf-8, which is firstly done to read the cfg:

```
  File "[...]\temp.py", line 5, in <module>
    rec.load("viewer/20210803_065512911_example_binary32.cff")

  File "[...]\Anaconda3\lib\site-packages\comtrade.py", line 760, in load
    self._load_cff(cfg_file)

  File "[...]\Anaconda3\lib\site-packages\comtrade.py", line 811, in _load_cff
    for line in file:

  File "[...]\Anaconda3\lib\codecs.py", line 322, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)

UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe8 in position 367: invalid continuation byte
```

I tried using the Version 265273d2dbb650c6dde11193d3f868edfa1d42fe, but that gives me another error:
```
File "[...]\Anaconda3\lib\site-packages\comtrade.py", line 1136, in parse
    values = row_reader.unpack(row)
TypeError: a bytes-like object is required, not 'str'
```
Here is an example file to reproduce the error:
[20210803_065512911_example_binary32.zip](https://github.com/dparrini/python-comtrade/files/6922823/20210803_065512911_example_binary32.zip)

dparrini, 2021-08-17:
Hi @chrisdinho , I pushed a new commit  (bbd3d72ad31b21f0d79fedf9c2313021eb5aba22) that may solve your issue. Let me know if you find any issue with the binary data parsing.

/David.

chrisdinho, 2021-08-19:
Hey @dparrini , huge thanks for your help!
That fixed my problem with this example. I tried it with a couple of other test files and got one which also produces an error:
`Traceback (most recent call last):

  File "[...]\temp.py", line 5, in <module>
    rec.load("viewer/20210217_173713922_example_float32.cff")

  File "[...]\Anaconda3\lib\site-packages\comtrade.py", line 778, in load
    self._load_cff(cfg_file)

  File "[...]\Anaconda3\lib\site-packages\comtrade.py", line 837, in _load_cff
    line = file.readline()

  File "[...]\Anaconda3\lib\encodings\cp1252.py", line 23, in decode
    return codecs.charmap_decode(input,self.errors,decoding_table)[0]

UnicodeDecodeError: 'charmap' codec can't decode byte 0x8f in position 749: character maps to <undefined>`

I attached the example file.

[20210217_173713922_example_float32.zip](https://github.com/dparrini/python-comtrade/files/7014451/20210217_173713922_example_float32.zip)

dparrini, 2023-09-09:
Fixed on release 0.1.1

closed 2023-09-09 by dparrini