dparrini/python-comtrade #19: Ability to export pandas dataframe to COMTRADE file

Episkiliski, 2021-08-13:
Hi,

great work with this package!

I just wanted to request a feature. Can you add the functionality to export a pandas dataframe to a COMTRADE file so I can read it with another software?

(I couldn't find this functionality).

Thanks!

dparrini, 2021-08-17:
Hi @Episkiliski , thank you for your request.

The possibility of creating/saving comtrade files was requested before. I'll address this and later make it easier to export from dataframes.

I'll keep this request open.

Episkiliski, 2021-10-02:
Great, thanks!

hildogjr, 2021-11-25:
Here is some related (limited) code https://github.com/relihanl/comtradehandlers/blob/master/src/writer.py that may be useful.

Episkiliski, 2022-12-21:
Hi @dparrini , I just wanted to check if there is any news in this development.

closed 2026-11-16 by dparrini