dparrini/python-comtrade #22: timestamp of REV_1991 should be different from REV_1999 and REV_2013

lkxajd, 2021-09-01:
timestamp of REV_1991 should be "month/day/year"

engineerjoe440, 2021-09-02:
Aha! Good catch! I've worked these suggestions into a [pull request](https://github.com/dparrini/python-comtrade/pull/25) #25 that we can review. I took a little liberty in the way that I created the conditional statement (since 1991 seems to be the only year that the order is different, and everything else is more consistent).

Let's see what @dparrini thinks, and if he is comfortable with the proposed changes, then he'll be able to merge the work!

Thank you for catching this!

dparrini, 2021-09-02:
Hi @engineerjoe440 , I merged your PR. Great job! @lkxajd thanks for your contribution, and let us know if the update solves the issue for you.

lkxajd, 2021-09-06:
I have tested the update, I think the issue has been solved.
Thanks.

closed 2021-09-06 by lkxajd