dparrini/python-comtrade #23: Default _rev_year should be REV_1991

lkxajd, 2021-09-01:
In "def _read_io",
```
        if 3 == len(packed):
            # only 1999 revision and above has the standard revision year
            self._station_name, self._rec_dev_id, self._rev_year = packed
            self._rev_year = self._rev_year.strip()

            if self._rev_year not in (REV_1991, REV_1999, REV_2013):
                if not self.ignore_warnings:
                    msg = WARNING_UNKNOWN_REVISION.format(self._rev_year)
                    warnings.warn(Warning(msg))
        else:
            self._station_name, self._rec_dev_id = packed
            self._rev_year = REV_1999
```
The default _rev_year is REV_1999, but it should be REV_1991.

engineerjoe440, 2021-09-02:
Hi @lkxajd!

It seems like you've been busy finding some issues! Thank you for reporting the items you've found. Perhaps @dparrini will better understand this, but I wanted to see if I could offer a hand and contribute as well.

I think I'm not very well understanding what your meaning is here. Could you elaborate on the following?
* What change *exactly* are you proposing?
* What is the purpose of the change (i.e., is something not working, is there something that is incorrect, etc.)?

I hope that this will help iron out exactly how we can help contribute to `python-comtrade`, thanks!

lkxajd, 2021-09-06:
- In IEEE C37.1 11 -1991 , the format of CFG file does not contain "rev_year", I think it's because the rev_1991 was the first revision.

[image]
- But in  IEEE C37.1 11 -1999 and 2013，it contains "rev_year".
[image]

- So, when we get of CFG file which has no "rev_year"，it should be "rev_1991". Maybe，the code I pasted wasn't correct, but I think the issue should be resoved.

dparrini, 2021-09-06:
Hi @lkxajd thanks for the clarification and @engineerjoe440 for the help. I pushed an update and will later release a new version.

closed 2021-09-06 by dparrini