dparrini/python-comtrade #26: CFG file with iso-8859-1 encoding or extend to other formats

vicnet, 2021-09-08:
Hello,
The Cfg class accept only simple ASCII or utf-8.
- ASCII is default value.
- _utf-8_ is detected automatically.

It should be great if one could choose the encoding format, other than these two.
By example, by passing `endoding=''xxx` in Comtrade constructor or in load method ?

dparrini, 2021-09-26:
Hi @vicnet,

Yes, I totally agree that it's great to have this option available. So I kept the current behavior as you listed and added the `encoding` parameter for all `load` methods we have as you suggested.

Try with the new version 0.0.9 changing `comtrade_obj.load('your_cfg_file.cfg', encoding='iso-8859-1') and let me know if it solves the issues you're having.

If you have available, and if it's possible, it would be nice to have an anonymous copy of the cfg file with the encoding you're currently having problems. I implemented a basic unit test for this case but I think it could be improved.

Thanks and let me know if I can close this.

vicnet, 2022-01-20:
Hello,

Thanks, for you patch.
It's working perfectly.

Regards,
Vicnet

closed 2022-01-20 by vicnet