dparrini/python-comtrade #28: error: iterative unpacking requires a buffer of a multiple of 14 bytes

StephTruong, 2021-11-28:
Trying to read a binary comtrade file composed of CFG file and DAT file with this code (opening the files on an online viewer works):

```
import matplotlib.pyplot as plt
from comtrade import Comtrade

rec = Comtrade()
rec.load("files/PGC10-20210717_180606_581.cfg", "files/PGC10-20210717_180606_581.dat",)
print("Trigger time = {}s".format(rec.trigger_time))

plt.figure()
plt.plot(rec.time, rec.analog[0])
# plt.plot(rec.time, rec.analog[1])
# plt.plot(rec.time, rec.analog[2])
# plt.plot(rec.time, rec.analog[3])

plt.legend([rec.analog_channel_ids[0], rec.analog_channel_ids[1], rec.analog_channel_ids[2], rec.analog_channel_ids[3]])
plt.show()

```

I am getting the error:

```error: iterative unpacking requires a buffer of a multiple of 14 bytes```

My cfg file looks like this:

```
PG10,PG,1999
3,3A,0D
1,IA,,,A,0.001,-2.048,0,-5000,5000,1,1,S
2,IB,,,A,0.001,-2.048,0,-5000,5000,1,1,S
3,IC,,,A,0.001,-2.048,0,-5000,5000,1,1,S
50
1
3200,64
17/07/2021,18:06:06.826009
17/07/2021,18:06:06.845009
BINARY
1
 ```

StephTruong, 2021-11-30:
My dat file had extra 2 bytes at the end of the file. Problem solved after truncating.

StephTruong, 2022-03-11:
Actually the values are inaccurate after truncating so it was a bad solution.

dparrini, 2023-09-10:
Hi, I'm closing this because I can't reproduce the issue without the .dat sample.

closed 2023-09-10 by dparrini