dparrini/python-comtrade #29: Clarification re. "partial" implementation of skew and multiple sample rates

drewsilcock, 2022-01-11:
Hi David, thanks for the useful library!

Would you be able to clarify the status of the "partial" implementation of the analog channel-specific skew and the multiple sample rates?

It seems from the code that while the skew is being read in from the cfg file, it isn't being used anywhere to calculate channel-specific timestamps via the `timestamp + (sample_number - 1) * skew` equation. Is this what is meant by "partial" implementation? If so, are there any plans to include the channel-specific skews? (I imagine it would require moving the timestamps from a global property to a channel-specific property.)

In terms of the multiple sample rates, from the code it looks like this is fully implemented - I can't see what edge cases wouldn't be correctly handled by the logic in the `DatReader._get_time()` method according to my understanding. Is there something I'm missing here?

Thanks in advance!

dparrini, 2022-01-12:
Hi @drewsilcock , thanks for your support!

To give you a quick answer, I'd say that every feature that I don't have any test or example data is something that has no or partial implementation - just because I can't be so sure if it works as intended.

In the case of the channel's skew I didn't come with any test case for this to check how much work is needed but you're probably right that it might not require much.

As you suggested, a possible implementation is moving the global timestamp logic into a local/per channel one. I think I'll have some challenges developing the interface to create dataframes from comtrade data work with time per channel, but I agree that it's the solution.

If you have any comtrade file that has this feature and don't mind to share ("anonymizing" it, removing descriptions, names and unneeded channels and data), please let me know.

drewsilcock, 2022-01-17:
Hi David, thanks for the quick response and the explanation.

I like the ethos that if it isn't fully tested, it isn't fully implemented - if only more people took this attitude!

I'm afraid I don't have any COMTRADE files with skews in but I'll be sure to let you know if I come across any.

dparrini, 2026-10-21:
Hi @drewsilcock, coming back to this one.

**Multiple sample rates:** you were right that the logic was almost there, but there was an edge case: `_get_time()` computed the time as `(n - 1) / samp` using only the rate of the segment the sample belongs to. Without timestamps in the DAT file (or when they are non-critical), every sample after a rate change got a wrong time, as if the whole record had been sampled at the new rate. The time is now accumulated segment by segment, and there are tests for ASCII and binary DAT files with two sample rates. The README now lists multiple sample rates as supported.

**Skew:** I've moved it to its own issue, #52. One note on the equation: the standard defines `skew` as the channel time skew (in µs) from the start of the sample period, so it's a constant offset per channel, `t_i(n) = t(n) + skew_i`, rather than something multiplied by the sample number.

The fix is in the `fix-issues-48-49-29` branch and will be in the next release. Closing this one. Thanks again!

closed 2026-10-21 by dparrini

dparrini, 2026-10-21:
Follow-up: these fixes are now released in [comtrade 0.2.0](https://github.com/dparrini/python-comtrade/releases/tag/0.2.0) (`pip install -U comtrade`), including the time skew support from #52 (`Comtrade.analog_time`).