dparrini/python-comtrade #30: Revision year for stadard IEC-2001

vicnet, 2022-01-20:
IEC standardizes COMTRADE 1999 only in 2001.
IEC-60255-24-Electrical-relays-Common-format-for-transient

I have some file with rev-year: 2001.
So the time factor is not read.

Is it possible to:
- add 2001 rev-year in comtrade module.
- add an equivalence in Comtrade constructor (2001=>1999)
- use version just below read rev-yead (2001 will set rev-year to 1999)
- other ideas ?

Here is a exrtact of the IEC std:
> rev-year - COMTRADE fileversion is defined by the year of the standard revision, e.g.
> 2000. Critical, numeric, minimum length = 4 characters, maximum length =
> 4 characters. This field will identify whether the file structure differs from the
> original C37.111-1991 COMTRADE standard. Absence of the field or an
> empty field is interpreted to mean that the file complies with the 1991 version
> of the standard

Regards,
Vicnet

vicnet, 2022-02-07:
Hello,
[0001-add-iec-60255-24-revision-2001.patch.txt](https://github.com/dparrini/python-comtrade/files/8015390/0001-add-iec-60255-24-revision-2001.patch.txt)

I have a patch that add 2001 as an available revision, in attached file.
This change the main file and doc file also...

I can push a pull-request my branche if you add me in contributors.

Regards,
Vicnet

dparrini, 2023-09-07:
Hi @vicnet sorry for the long wait. This IEC doc is new for me. Since its an standard, I guess I must add it to the module. I expect that anyone can contribute here. Let me know if you can create pull requests.

dparrini, 2023-09-09:
Revision 2001 added at commit e5163ed or release 0.0.13.

closed 2023-09-09 by dparrini