dparrini/python-comtrade #37: 999999 == Null

tieugene, 2022-09-10:
According to 1991/1999 standards sample value `999999` means "no signal'

engineerjoe440, 2022-09-13:
@tieugene, could you elaborate on what you're suggesting? Were you thinking that `python-comtrade` should set such values to a Python `None`? I think I remember that mentioned in the standard, would you be able to point out the section where it's referenced?

tieugene, 2022-09-13:
> @tieugene, could you elaborate on what you're suggesting? Were you thinking that `python-comtrade` should set such values to a Python `None`? I think I remember that mentioned in the standard, would you be able to point out the section where it's referenced?

- IEEE C37.111-1991, "6.3.4 Format of Data Values":
"The data values shall be represented in the integer format of six digits (I6) separated by commas. Missing values shall
be represented by 999999. Status information (I1) shall be represented by ones and zeroes"

- IEEE C37.111-1999, "6.3. "ASCII data file format":
"...minimum value = Ð99999, maximum value = 99998. Missing analog values must be represented by placing the value 99999 in the field."

- IEEE C37.111-2013, "8.4 ASCII data file format":
"minimum value = -3.4028235E38, maximum value = 3.4028235E38. Missing analog values must be represented by data
separators immediately following each other with no spaces (null)"

tieugene, 2022-09-20:
> @tieugene, could you elaborate on what you're suggesting? Were you thinking that `python-comtrade` should set such values to a Python `None`? I think I remember that mentioned in the standard, would you be able to point out the section where it's referenced?

BTW, value `''` (empty string) means 'no signal' too.
An example:
```
--- file type: DAT ASCII ---
1,0,60,-44,-15,0,0,0,0,-12704,9092,3627,-21934,21931,-1925,61,-44,-14,-5,,1,1,0,0
2,833,59,-24,-35,0,0,0,0,-11692,5796,5953,-20190,20268,-2464,60,-24,-34,-5,,1,1,0,0
```

dparrini, 2023-09-07:
Hi all, thanks for the discussion and pointing this issue. This one is tricky since the missing data value varies with revision year and format, but I managed to implement a solution.

It's not exactly as suggested (using `None`) but using `NaN` values instead. The reason being that internally analog values are stored on `array.array` or `numpy.ndarray` that don't allow mixed types.

Please let me know if I can close this one.

closed 2023-09-10 by dparrini