dparrini/python-comtrade #41: Should this Project Fork or Become Managed by a Group?

engineerjoe440, 2023-01-16:
Hi @dparrini, and community!

I want to offer an open conversation about whether this project should fork, or move to group maintainer-ship to allow splitting out the work of maintaining, accepting pull-request, responding to issues, and providing new releases. I would like to get your thoughts, @dparrini, since this is originally your work, and I don't wish to make this feel as though I'm suggesting that this project be taken away from you. You've got a really awesome project, and its clear you've put a lot of time, energy, and heart into the project.

I wonder if the project's slow-down is due to other factors that have just kept you from being active with it. I know that's something that comes up a lot for developers (myself, included). So I wonder if sharing some of this work would be helpful to you?

I know that I would be willing to jump in and offer my hand where I can. I'd be happy to work alongside you as a maintainer. Alternatively, if you're no longer interested in maintaining this project, I'd be happy to help the community sustain a fork where possible.

Thank you so much for your work on this project, and again, I want to emphasize that this is something I'm interested in helping you wherever possible. I really would hate to "steal" your project from you. That is not my goal, in any way.

dparrini, 2023-01-16:
Hi @engineerjoe440 , thanks for your suggestion. I've been busy due to job exchange and moving to another country, so this project hasn't been maintained for a while. I see some open issues and pull requests but didn't have the time to check all of them. I hoped that I'd have some time for a refactor, to add the feature to write Comtrade files, and checking out GitHub CI's features (as your previous PR), but it wasn't the case.

It also doesn't help that I don't work directly with comtrade files or their analysis anymore, so in the near feature I might miss new versions of the standard and its real world usage.

Having said that, I think by the end of this month I'll find some time to figure this out but until then I think we can leave this discussion open. People to contribute are always welcome and I'm happy that during all the time this repo has been inactive only few issues poped up and some users where able to contribute with fixes. I don't discard having another maintainer alongside as well for the reasons above.

Thanks,
David.

engineerjoe440, 2023-01-16:
@dparrini, I can sympathize! I hope that through all of your life's recent changes that you've remained well, both physically, and mentally. I understand that moving and changing roles in career can be overwhelming to say the least. So I do not want to diminish that impact! I think that you've really put together some terrific work here, and I want to thank you for that!

I wonder if it may be helpful to utilize a GitHub organization to allow multiple project maintainers?
[image]

Again, I'm not sure whether this is exactly what you're interested in, but it seems like it may be a path towards having multiple folks support the project as co-maintainers.

Thank you again, David!
~Joe

closed 2026-10-21 by dparrini