dparrini/python-comtrade #42: Patch for DAT file extension be case sensitive

asierllano, 2023-02-15:
The automatic .dat automatic filename are always lowercase even for .CFG files which are uppercase.
This is a problem for case sensitive file systems (like linux).

I have provided a patch that makes use .DAT files for .CFG files and .dat files for .cfg files.
This makes my whole database of non-coherent extension comtrade files work in linux.

Patch provided with the fix:
[comtrade_extension_case.patch.gz](https://github.com/dparrini/python-comtrade/files/10741327/comtrade_extension_case.patch.gz)

tieugene, 2023-02-15:
It is not impossible that extension can be .cfg, CfG, Cfg etc.
And .dat, .DaT, dAt, .Dat etc.

asierllano, 2023-02-15:
Yes. That is true.

You always have the manual specification of the filename.
I just wanted the automatic process to solve the most common thing is that
both files are ".cfg" and ".dat" or ".CFG" and ".DAT". Do you think that it
is useful another more complex solution. I have many COMTRADE files from
different sources and vendors and I have all of them one way or the other.
I don't have any other combination.

Best regards,
Asier

tieugene, 2023-02-16:
I think it must be more steady case-insensivitive solution to handle as extension as filename itself.
It is not impossible to have `something.cfg` and `SOMETHING.DAT` in pair.

asierllano, 2023-02-16:
OK. So as you suggest, here is the case insensitive patch.

dparrini, 2023-09-07:
Hi,
There are multiple combinations indeed. But since the module provides this automatic file pairing feature I agree that it would be nice for it to work in these cases.

In the updated version the module will look for matching cases for each extension's characters. So `.cfg` => `.dat`, `.Cfg` => `.Dat`, `.CFG`=> `.DAT`, and so on.

If it doesn't follow this rule, the extra and optional positional argument should be used instead as in the test samples:

```python
comtrade = Comtrade()
comtrade.load("tests/sample_files/sample_bin.cfg", "tests/sample_files/sample_bin_missing.dat")
```

closed 2023-09-07 by dparrini