dparrini/python-comtrade #44: How to extract phase information from the comtrade files?

SidhartheneeNayak, 2023-07-28:
[image]
The .cfg and .dat files when analyzed in Wavewin software show the phase column as well. How to extract this information using this Comtrade package in Python?

tieugene, 2023-07-28:
> The .cfg and .dat files when analyzed in Wavewin software show the phase column as well. How to extract this information using this Comtrade package in Python?

Use Fourier transform

SidhartheneeNayak, 2023-07-28:
@tieugene can you please elaborate? I am able to get only the instantaneous time series values of all analog channels. Is there a way to get instantaneous phase time series values of all analog channels?

tieugene, 2023-07-28:
> @tieugene can you please elaborate? I am able to get only the instantaneous time series values of all analog channels. Is there a way to get instantaneous phase time series values of all analog channels?

Yes, of course.
You can apply FFT for signal series and get complex value of signal.
Then get 1st harmonic etc.
Note: the nuance is that "electricians" use FFT informally; look at [this](https://github.com/tieugene/iosc.py/blob/main/iosc/core/sigfunc.py) (`hrm1(...)`)

SidhartheneeNayak, 2023-07-30:
@tieugene In the link you have given above,  in the _sft function, 'a' is the time series signal but what does 'n' and 'w' indicate? Suppose from a disturbance record, I extract the time series value of phase A current whose length is 1000. Then 'a' is an array of size 1000 but what are 'n' and 'w'?

tieugene, 2023-07-31:
> @tieugene In the link you have given above, in the _sft function, 'a' is the time series signal but what does 'n' and 'w' indicate? Suppose from a disturbance record, I extract the time series value of phase A current whose length is 1000. Then 'a' is an array of size 1000 but what are 'n' and 'w'?

`n` is starting index of an array, what we calculate from. `w` - 'window' == signal period in points. E.g. for signal 50Hz and sample rate 1200Hz `a`=60. Than if we need to know harmonics in point 2s, than w=60, n=1200*2-60

SidhartheneeNayak, 2023-07-31:
> `n` is starting index of an array, what we calculate from. `w` - 'window' == signal period in points. E.g. for signal 50Hz and sample rate 1200Hz `a`=60. Than if we need to know harmonics in point [...]

In this why is a=60? Also, how to find the harmonic at each time step up to 2 seconds and not just at 2 seconds?

tieugene, 2023-07-31:
> In this why is a=60? Also, how to find the harmonic at each time step up to 2 seconds and not just at 2 seconds?

Sorry, I'm wrong: `w`=1200/50=24.
`a` is an array (whole of array), it cannot be 60.
To find harmonics of each point you have to call this function for each point.
Excluding starting `w` points.

SidhartheneeNayak, 2023-07-31:
Ok and if we take the angle of the complex value of the harmonics at each time step then we will get the phase value, right?

tieugene, 2023-07-31:
> Ok and if we take the angle of the complex value of the harmonics at each time step then we will get the phase value, right?

Right. You can get real/imag value of couples or module/angle, for you choice

closed 2024-03-11 by dparrini