dparrini/python-comtrade #45: Question about time

fabrix76, 2023-10-06:
Hello David,
I'm approaching to analyze comtrade files using python and I have som dubs regarding time.

Looking row ad my .dat file, i see, in secon coloumn (timestamp) a progressive value incrementing always of the same quantity (625)
and I have 6405 samples.
Using library and loading data into dataframe i see that value became 0,000625 and last value is 4,002500 
In .cfg file i can see start and end time, I undestand it is the time window of the sample (1 second of window in my case), so my sample rate shoud be 1 sec / 6405 samples and I was thinking that it correspond to each line increment (625) but it's wrong.

Can you help me to find where I'm wrong?

thank you in advance, for you good job and for the support

Regards

xiiaochuang, 2025-06-05:
Hi there,

I have the same problem. After check through the code, I figure the time calculation does not adjust to comtrade file with multiple sample rates. It calculates timestamp using the first sample rate.

https://github.com/dparrini/python-comtrade/blob/fd8a60fba591f8aebcd849ed6b1585247bdfd95d/comtrade.py#L1153-L1154
To fix it, I use the following preprocess:

```python
rec = comtrade.load(...)

time = []
val = 0
endsamp_ = 0
for samp, endsamp in rec.cfg.sample_rates:
    if endsamp > endsamp_:
        for i in range(endsamp_, endsamp):
            val += 1/samp
            val = round(val, 7)  # Handle precision loss for float numbers
            time += [val]
    endsamp_ = endsamp
```

dparrini, 2026-11-16:
Hi, sorry for the very late reply.

Your reading of the DAT file is right, the misunderstanding is about the two CFG timestamps. They are not the start and end of the recording:

- the first one is the time of the **first sample** of the record;
- the second one is the **trigger** time (the event that started the recording), which usually happens shortly after the first sample, since recorders keep some pre-fault data.

So the record length isn't given by the CFG timestamps, but by the samples themselves. The DAT timestamps are in microseconds (multiplied by the CFG `timemult` field), so an increment of 625 means 625 µs between samples, i.e., a sample rate of 1 / 625 µs = 1600 Hz. With 6405 samples, the last one is at (6405 - 1) × 625 µs = 4.0025 s, which is exactly what the library returns in `Comtrade.time`. You can confirm the sample rate in the CFG file sample rate line (`samp,endsamp`, e.g., `1600,6405`), also available in `rec.cfg.sample_rates`, and the relative trigger time in `rec.trigger_time`.

Closing this one, but feel free to reopen it if something still doesn't add up.

closed 2026-11-16 by dparrini