dparrini/python-comtrade #47: How to read the contents of .cfg file

SidhartheneeNayak, 2024-03-11:
A disturbance record consists of .cfg and .dat files.  But I to read line by line the contents of the .cfg file using the comtrade package. Can anyone please tell which function to use to read line by line the contents of a .cfg file. I have attached a screenshot of how a .cfg file looks when opened in notepad.
[image]

dparrini, 2024-03-11:
Hi @SidhartheneeNayak , on our [readme we have an example on how the comtrade files can be read](https://github.com/dparrini/python-comtrade#cfg-and-dat-files-all-revisions). The module always loads both .cfg and .dat file. Do you want to read only the cfg contents? If that's the case, it should be possible with internal module's features. Let me know if it's the case.

Here's a more detailed example with the data you can extract from the comtrade cfg file:

```python
import comtrade

rec = comtrade.load("sample_files/sample_ascii.cfg", "sample_files/sample_ascii.dat")

print(rec.cfg.file_path)
print(rec.cfg.station_name)
print(rec.cfg.rec_dev_id)
print(rec.cfg.rev_year)
print(rec.cfg.analog_channels)  # list with analog channel definitions
print(rec.cfg.status_channels)  # list with status channel definitions
print(rec.cfg.time_base)
print(rec.cfg.frequency)
print(rec.cfg.ft)  # dat file format
print(rec.cfg.timemult)
print(rec.cfg.start_timestamp)
print(rec.cfg.trigger_timestamp)
print(rec.cfg.nrates)
print(rec.cfg.sample_rates)  # list with sample rates
```

SidhartheneeNayak, 2024-03-21:
Thanks for the information I was able to read the contents of only the cfg file. I wanted to know if there is a way by which we can read the time series data of the digital binary channels just like we read the time series data of currents and voltages from the analog channels.

closed 2024-04-13 by dparrini